embeddedTS/linux-lts #19: i.MX6: wl12xx documentation says it should use vwlan-supply

ts-kris, 2023-03-20:
Currently, we're not using that. Though nor is TIs reference board in-kernel. The TiWi is showing issues with cycling the interface up and down which may be partly related to not having that property defined. Need to look closer at this.

markfeathers, 2023-03-20:
Worked around it with the below patch, but this has not been tested on silex:

```
diff --git a/arch/arm/boot/dts/imx6qdl-ts7970.dtsi b/arch/arm/boot/dts/imx6qdl-ts7970.dtsi
index 56671bf36ce2..bc6b36c8511b 100644
--- a/arch/arm/boot/dts/imx6qdl-ts7970.dtsi
+++ b/arch/arm/boot/dts/imx6qdl-ts7970.dtsi
@@ -96,7 +96,7 @@ reg_usb_otg_vbus: regulator-usb-otg-vbus {
                enable-active-high;
        };
 
-       reg_wlan_vqmmc: regulator-wlan-vqmmc {
+       reg_wlan_vmmc: regulator-wlan-vmmc {
                compatible = "regulator-fixed";
                regulator-name = "WLAN_1V8";
                regulator-min-microvolt = <1800000>;
@@ -651,8 +651,7 @@ &usdhc1 {
        pinctrl-0 = <&pinctrl_usdhc1>;
        pinctrl-1 = <&pinctrl_usdhc1_100mhz>;
        pinctrl-2 = <&pinctrl_usdhc1_200mhz>;
-       vmmc-supply = <&reg_3v3>;
-       vqmmc-supply = <&reg_wlan_vqmmc>;
+       vmmc-supply = <&reg_wlan_vmmc>;
        bus-width = <4>;
        fsl,tuning-step = <2>;
        cap-sdio-irq;

```

ts-kris, 2023-06-29:
vwlan-supply is for SPI modes, not SDIO.

closed 2023-08-17 by markfeathers