embeddedTS/linux-lts #30: Debian 12 Ethernet Issue

jdeinum, 2023-08-09:
I am running into some issues with Ethernet with the most up to date 5.10.y kernel. It works on the stock `debian-armhf-bookworm-headless-latest.tar.bz2` image, but if I compile the kernel as demonstrated [here](https://docs.embeddedts.com/TS-4900#Linux_5.10.y), then I can no longer get a DHCP lease.

# Steps to Reproduce

## Compiling the Kernel
I use the standard `make tsimx6_defconfig`. Other than adding a few new kernel modules (some gpio stuff), and adding another device tree, I do not touch the kernel. The device tree just adds another SPI device. This kernel is then moved to the root filesystem (in this case, debian12-headless)

## Booting
The standard sdboot is used. Device boots fine, and I am able to login into root.

## Checking Ethernet Status
Here is all the info for the new image:

```
root@tsimx6:~# uname -r
5.10.170-00066-gdeecb184ff05
root@tsimx6:~# readlink /sys/class/net/end0/device/driver
../../../../../bus/platform/drivers/fec
root@tsimx6:~# dhclient -v end0
Internet Systems Consortium DHCP Client 4.4.3-P1
Copyright 2004-2022 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

[   86.088666] Micrel KSZ9031 Gigabit PHY 2188000.ethernet-1:07: attached PHY driver [Micrel KSZ9031 Gigabit PHY] (mii_bus:phy_addr=2188000.ethernet-1:07, irq=POLL)
Listening on LPF/end0/00:d0:69:53:2c:54
Sending on   LPF/end0/00:d0:69:53:2c:54
Sending on   Socket/fallback
DHCPDISCOVER on end0 to 255.255.255.255 port 67 interval 4
DHCPDISCOVER on end0 to 255.255.255.255 port 67 interval 10
[   92.325281] fec 2188000.ethernet end0: Link is Up - 1Gbps/Full - flow control rx/tx
[   92.333006] IPv6: ADDRCONF(NETDEV_CHANGE): end0: link becomes ready
DHCPDISCOVER on end0 to 255.255.255.255 port 67 interval 12
DHCPDISCOVER on end0 to 255.255.255.255 port 67 interval 14
DHCPDISCOVER on end0 to 255.255.255.255 port 67 interval 10
DHCPDISCOVER on end0 to 255.255.255.255 port 67 interval 11
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
```
Not able to get a lease in this case. And the info for the stock image:
```
root@tsimx6:~# uname -r
5.10.170
root@tsimx6:~# readlink /sys/class/net/end0/device/driver
../../../../../bus/platform/drivers/fec
root@tsimx6:~# dhclient -v end0
Internet Systems Consortium DHCP Client 4.4.3-P1
Copyright 2004-2022 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/end0/00:d0:69:53:2c:54
Sending on   LPF/end0/00:d0:69:53:2c:54
Sending on   Socket/fallback
DHCPDISCOVER on end0 to 255.255.255.255 port 67 interval 8
DHCPDISCOVER on end0 to 255.255.255.255 port 67 interval 17
DHCPOFFER of 10.60.32.159 from 10.60.32.1
DHCPREQUEST for 10.60.32.159 on end0 to 255.255.255.255 port 67
DHCPACK of 10.60.32.159 from 10.60.32.1
bound to 10.60.32.159 -- renewal in 14238 seconds.
```

In this case the DHCP lease works as expected. Not entirely sure what the problem is, but I know there was a fix released for this not too long ago. The fix solved it for the debian 11 system but doesn't seem to fix it for debian 12. 

I hope I didn't miss anything obvious.

markfeathers, 2023-08-09:
This sounds like the [skew issue](https://github.com/embeddedTS/linux-lts/pull/25) that was recently merged in. The stock images we just generated included this as a fix.

From your kernel:
```
root@tsimx6:~# uname -r
5.10.170-00066-gdeecb184ff05
```

The end string -g means git, and the rest after that is part of the hash of the last commit "deecb184ff05".  It looks like this doesn't match any of our commits after that skew merge:
https://github.com/embeddedTS/linux-lts/commits/linux-5.10.y

I'm assuming you just had some modifications on top of our kernel, but can you try building the stock kernel and make sure it has on of our hashes (eg, 9311369 right now)?  With the latest kernel on debian 11 or 12 I'm currently not reproducing any issues.

jdeinum, 2023-08-09:
Ya I put some of our own stuff in the kernel and just made a commit to get rid of the `dirty` suffix :) . I will use the stock kernel in the morning and put an update here.

jdeinum, 2023-08-09:
I definitely think this is it. I overwrite the `arch/arm/boot/dts/imx6qdl-ts4900.dtsi` with a slightly modified version, and don't include that fec property. I'll assume this fixes it, so I will close this issue for now. Thanks for getting back to me so quickly.

closed 2023-08-09 by jdeinum