embeddedTS/linux-lts #32: TS-7250-V3: gpiochip#s out of order depending on model

markfeathers, 2023-08-24:
On the latest 5.10 kernel the gpiochip devices are out of the documented order on the high end model. The higher end model includes the i2c gpio expander which loads before the FPGA GPIO. This should be loaded in gpiochip order and a regression test added to make sure this remains consistent.

ts-kris, 2023-08-24:
I'm not sure if the kernel can be coaxed to load modules in a specific order unless we make one a builtin. I feel like this is a distribution problem, especially since the `base` FDT property is long deprecated. Is there another way to do this in the kernel that I'm missing?

I think our options here are:
- Make the FPGA GPIO driver a builtin.
- Document it. Give the chip a name along with its pins as we have been, and refer to them by that in the manual. At the very least, note that the actual I/O chip may change depending on which model is used.
- Fix it in `distro-seed` and/or `buildroot` to load the modules in a specific order via `/etc/modules.d` or `/etc/init.d/` scripts.

Thoughts?

MarkFeatherston, 2023-08-29:
I had thought the aliases would handle this:
https://github.com/embeddedTS/linux-lts/blob/d38569efbcd84f265f2e2371993a69eb0fb2c7e0/arch/arm/boot/dts/imx6ul-ts7250v3.dtsi#L30

It looks like we build the driver in:
https://github.com/embeddedTS/linux-lts/blob/d38569efbcd84f265f2e2371993a69eb0fb2c7e0/arch/arm/configs/tsimx6ul_defconfig#L562C20-L562C20

That same config has CONFIG_GPIO_PCA953X=y which is the problematic bank that beats our FPGA I/O.  I'm tempted to make that a module in our defconfigs to work around this to make the transition a bit simpler since some users are hitting this, but in the future we might need to use bank names in our documentation if aliases do not guarantee an order.

ts-kris, 2023-08-29:
I think aliases for GPIO only apply to the FDT naming itself unless you have testing that shows otherwise.

I'd agree that moving PCA953X to a module would likely be the best way to guarantee ordering. I think bank names are what the kernel really wants you to use these days, but I'm not up to speed on the GPIO subsystem. Might be worth digging through some kernel docs.

closed 2023-09-14 by markfeathers