embeddedTS/linux-lts #51: TS-7100: 5.10.y: Verify GPIO labeling

ts-kris, 2023-10-23:
In testing, I've found that gpioinfo reports relays on `gpiochip2` however the actual GPIO numbers for them are `4 4` and `4 5`. Need to re-verify all GPIO labels to ensure they are correctly aligned with real GPIO.

lionel, 2023-11-20:
Since the beginning (~2019), the TS-7100 device tree had been using the alias `&gpio3` for GPIO5 rather than `&gpio5`, and had been starting the FPGA aliases with two that collide with CPU aliases (`&gpio4` and `&gpio5`). Since using the chip/line numbers directly works, it is possible that it has always been broken, or that the assignment of `&gpio` aliases to `gpiochip` numbers changed in 5.10.

Either way, a P.R. is being created with the fix for this problem along with some problems that were found while investigating it.

ts-kris, 2023-12-21:
Not sure why this wasn't closed automatically, but this looks like it was fixed in #56

closed 2023-12-21 by ts-kris