embeddedTS/linux-lts #117: Add CONFIG_IIO_SYSFS_TRIGGER to 6ul defconfigs

markfeathers, 2024-08-16:
Test with example high speed ADC:
https://community.nxp.com/t5/i-MX-Processors/An-adc-using-an-imx6ull-processor-enables-continuous-acquisition/m-p/1719805

ts-kris, 2024-08-17:
Looks like this is a module in both 5.10 and 6.6, `iio-trig-sysfs`

Do we want to make it a builtin, or just document its use?

markfeathers, 2024-09-09:
Looks like I missed that, but we should test & document to get the higher speed ADC support.

markfeathers, 2024-10-02:
Took a quick swing at this while testing another i.MX6ul ADC issue.  This is sort of working, but has a surprising side effect.  It only supports a single channel.

```
# Add trigger device
modprobe iio-trig-sysfs
echo 0 > /sys/bus/iio/devices/iio_sysfs_trigger/add_trigger
echo "206250" > /sys/bus/iio/devices/iio\:device2/in_voltage_sampling_frequency
iio_readdev -t trigger0 -s 1024 -b 1024 iio:device2 voltage0 > samples.bin
```

The iio interface should permit us to do this:
```
iio_readdev -t trigger0 -s 1024 -b 1024 iio:device2 voltage0 voltage1 voltage5 voltage8 voltage9 > samples.bin
```
But this fails with:
```
Unable to allocate buffer: Invalid argument (22)
```
Ultimately right now we cannot enable multiple channels as scan elements and I'm not sure why:
```
root@tsimx6ul:~# echo 1 > /sys/bus/iio/devices/iio\:device2/scan_elements/in_voltage0_en 
root@tsimx6ul:~# echo 1 > /sys/bus/iio/devices/iio\:device2/scan_elements/in_voltage1_en 
-bash: echo: write error: Invalid argument
```